tessel/tessel-rust #3: Provide files Rust 1.0 stable for easy-getting-started experience

flaki, 2015-09-08:
Provide at least Linux, and ideally Mac builds for at least Rust 1.0 stable, explore whether latest (1.2) would be better. Also work with #1 to provide better instructions/tools for users to build their own SDK/standard lib for the latest Rust version.

johnnyman727, 2015-09-10:
@flaki is there an easy way to bundle the Rust compiler with the Tessel 2 CLI?

flaki, 2015-09-15:
I imagine there could be, but since the CLI is JS (cross-platform), while Rust (though, *is* cross-platform) sports different binaries for different systems it might not be worth it.

We could, rather provide it as an option and install it while installing the CLI.

We could also do "dev packages", in which we bundle all needed pre-requisites (compiler, virtual machine, scripts that make all the steps easier) into a single download, but that's a much bigger endeavor.

kteza1, 2015-09-15:
```
We could, rather provide it as an option and install it while installing the CLI.
```

This sounds good. Or else you need to package multiple rust comilers (one per platform)

johnnyman727, 2015-09-15:
Another option would be to have a "compilation server" that can accept a tarball and return a binary. It would only be used if the CLI couldn't find a rust compiler on the users PATH. That way, serious Rust developers can use whatever compiler they download but new Rust users can make use of an out of the box solution (assuming they have Internet access).

kteza1, 2015-09-15:
@johnnyman727 Hi Jon. But how fast is compilation server? Rust build system will bootstrap native toolchain. In my desktop, this takes around 1 hour. Then that toolchain will be used to cross compile all necessary crates including `std` for target (mips in case of tessel 2). So how fast the installation goes depends on the speed of compilation server.

johnnyman727, 2015-09-17:
@kteza1 It takes you an hour to compile any Rust project? I think it takes a long time to build Rust itself but only a few seconds (or less) to compile any given project.

kteza1, 2015-09-17:
@johnnyman727 I totally forgot that this is a one time task. you are right.

johnnyman727, 2016-01-01:
Just for the sake of logging everything I'm working on, you can find a `1.0.0` build for a Linux VM [here](http://static.rust-lang.org/dist/rust-1.0.0-x86_64-unknown-linux-gnu.tar.gz), the OpenWRT SDK [here](https://kevinmehall.net/tmp/OpenWrt-SDK-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2) and a pre-built `std` lib (specifically for 1.0.0) [here](https://www.dropbox.com/sh/e5fy42812q4am68/AAAT0-jyJN8FHeUtabQkXkwda/1.0.0/rust-2015-05-14-a59de37-mips-unknown-linux-gnu-4ec0bbe33bf2c3b7beff9ad0bc4d399a2d287b47.tar.gz?dl=0).

I compile files like this (in this case, my file is `test.rs`):
```
rustc -L ../mips-unknown-linux-gnu/lib --target=mipsel-unknown-linux-gnu -Ctarget-cpu=mips32r2 test.rs -Clinker=mipsel-openwrt-linux-gcc -O
```

I took a few stabs at compiling my own standard libs but didn't succeed today (just tried [this project](https://github.com/japaric/rust-on-openwrt) but have not tried @kevinmehall's guide [here](https://gist.github.com/kevinmehall/16e8b3ea7266b048369d). I hope to be able to get the std libs compiling for the latest (1.5.0) soon.

I do think that distributing a compiler and toolchain to everyone would be a huge headache. We have to match the compiled standard libs with whatever version of rust they're using. I think it would be easier to just have one copy of each release on a server, and send the rustc version along with the source code to the server to be compiled...

johnnyman727, 2016-01-01:
I'll also try out https://github.com/tessel/t2-firmware/tree/master/scripts/port_test#rust-cross-toolchain tomorrow.

flaki, 2016-01-01:
Whoah this sounds awesome!
I had some conversations at last month's Mozlando about which version should we be basing Rust support around and got an unequivocal 1.5 as an answer, but any progress here is great! (TBH I was tied up with other stuff, and now am working on the T2 AP+Arduino programmer base project for FOSDEM).

I'd love to look into this more - I feel the x-compiling story is strong with the T2, we sould be working on easing & documenting that workflow.

Also the T2+Rust interaction will be huge once I get around finding a way for making @dherman's [Neon](https://github.com/dherman/neon) working on/for the T2; that would be absolute *fractal*-ish carnage right there... :blush:

johnnyman727, 2016-01-01:
> Also the T2+Rust interaction will be huge once I get around finding a way for making @dherman's Neon working on/for the T2; that would be absolute fractal-ish carnage right there... :blush:

Indeed, Neon inspired me to dig back into Rust in the first place! I'm currently trying to figure out how to cross compile just the standard lib so I don't have to wait for hours each time.

kevinmehall, 2016-01-01:
If you're using a stable release of Rust, you won't be able to compile `libstd`, because it depends on unstable features and compiler-internal APIs that are disabled (but not removed) in stable release builds. One approach is to download the nightly corresponding to the stable release.

There is a hack that can re-enable those features in release builds, and I'm about to try implementing it in my `build-rust-libs.sh` script.

Also, I don't know what the state of the Tessel openwrt build and update infrastructure is, but it should be made to build and host the OpenWrt SDK tarballs. No guarantees how long anything will remain in https://kevinmehall.net/tmp/.

johnnyman727, 2016-01-01:
> Also, I don't know what the state of the Tessel openwrt build and update infrastructure is, but it should be made to build and host the OpenWrt SDK tarballs. No guarantees how long anything will remain in https://kevinmehall.net/tmp/.

Good point. I moved the one version of the SDK we have to Tessel S3 and updated the link in my above comment.

closed 2016-09-02 by Frijol