tessel/tessel-rust #5: Library claims LED brightness files don't exist on Tessel

johnnyman727, 2016-05-01:
When I cross-compile this library for Tessel and run the hello world example, I get the following error I create a new Tessel:
```
thread '<main>' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 2, message: "No suc
h file or directory" } }', src/libcore/result.rs:738
```

It is being thrown from [this line](https://github.com/tessel/tessel-rust/blob/master/src/lib.rs#L94). I have confirmed that the expected path does exist on Tessel so I'm not sure what to look at next.

johnnyman727, 2016-05-01:
Here is the binary if anyone wants to run it themselves on their T2.

kevinmehall, 2016-05-01:
`/sys/devices`, not `/sys/device`

johnnyman727, 2016-05-01:
I owe you a six pack of root beer for such a dumb question. 🍻

commit 00bffaae (2016-05-01): Fix sysfs LED path. Fixes #5
commit d7bbeb60 (2016-05-01): Merge pull request #6 from tessel/km-fix

Fix sysfs LED path. Fixes #5
closed 2016-05-01 by johnnyman727