tessel/tessel-rust #7: Readme needs an update

Frijol, 2016-05-17:
* [ ] GPS module is listed under USB
* [ ] Setup step 1 should probably link to Rust's own install recommendations
* [ ] Switch from Vagrant to a native compiler?
* [ ] Contact shouldn't link to Ken's email
* [ ] Contact should direct people to the rust channel on slack

cflewis, 2016-06-14:
Where is this deployments discussion taking place? All the Vagrant stuff seems like a pretty harsh learning curve :(

cflewis, 2016-06-15:
The README does not currently work. The rust libraries are different versions I think  (the Readme alludes to a linker issue and handwaves a "do it yourself" which isn't helpful :/ )

I'm going to try following a workflow which seems to be building for OpenWRT on Travis CI tonight, but as it stands, I think it's a bit premature for the Tessel Web site to advertise rust. It's not possible for the target Tessel audience to use it right now.

johnnyman727, 2016-06-16:
For the sake of anyone watching this issue, I recommend using [this Docker build](https://github.com/tessel/rust-compilation-server) instead of trying to get Vagrant linking working.

Frijol, 2016-09-02:
@johnnyman727 should this be closed, or relabeled?

closed 2016-09-02 by johnnyman727