tessel/tessel-rust #41: ServoArray constructor

tcr, 2016-09-17:
Right now we pass in two false flags for the I2C constructor, since we control two address bits.

Is there a way we can:

- Do the normal thing by default
- Optionally pass in what the I2C ID *should* be
- Restrict it to IDs that we can support in a type-safe way

rwaldron, 2016-09-19:
Side question: Why is this called ServoArray? It doesn't appear to be an array Servos...?

tcr, 2016-09-20:
@rwaldron You'll want to read... #40 ;)

Dr-Emann, 2016-09-20:
It might be helpful to link to the specific line(s) you're referring to.

tcr, 2016-09-20:
The code in question is the [ServoArray::new](https://github.com/tessel/tessel-rust/blob/master/servo-pca9685/src/lib.rs#L39) constructor, which sets the two GPIO address lines which modify the PCA9685 address itself.

Dr-Emann, 2016-09-21:
The usual way to do something like this in rust would be to have two functions:
`ServoArray::new()` and `ServoArray::with_something(something: Something)`. 

I'm not sure how those bits should be used, is it something like `address = I2C_ID | (addr2 << 2) | (addr3 << 3)`? (assuming addr2/3 are 0/1 at that point)

Where would the user find information about what those bits should be for their particular device? That might determine what they should pass in (e.g, if they would know the full address, that's probably what should be passed, but panic if it doesn't match what we expect)