tessel/tessel-rust #59: Can't build OpenSSL-dependent crates

felixalias, 2016-11-24:
Hey all,
I used to be able to build a project that uses crates like `hyper` which depend on OpenSSL (and a C compiler) with remote compilation (Aug 14th, at least). After upgrading to the newer versions of the CLI, this fails when trying to invoke the compiler:

```
[~/Desktop/Current/code/rust/t2quickstart]$ t2 run blinky --rustcc
INFO Looking for your Tessel...
INFO Connected to Sunfury.
INFO Compiling Rust code remotely (--rustcc)...
ERR! Error: Downloading hyper v0.9.10
ERR!  Downloading libc v0.2.15
ERR!  Downloading lazy_static v0.2.1
ERR!  Downloading gcc v0.3.34
ERR!    Compiling matches v0.1.2
ERR!    Compiling language-tags v0.2.2
ERR!    Compiling log v0.3.6
ERR!    Compiling cfg-if v0.1.0
ERR!    Compiling semver v0.1.20
ERR!    Compiling rustc_version v0.1.7
ERR!    Compiling lazy_static v0.2.1
ERR!    Compiling pkg-config v0.3.8
ERR!    Compiling openssl-sys v0.7.17
ERR! error: could not exec the linker `cc`: No such file or directory (os error 2)
ERR!   |
ERR!   = note: "cc" "-Wl,--as-needed" "-Wl,-z,noexecstack" "-m64" "-L" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "/tmp/tessel-rust-compile1161024-1615-1nkkefz.34qfzjjor/t2quickstart/target/release/build/openssl-sys-a50462f4200b258f/build_script_build.0.o" "-o" "/tmp/tessel-rust-compile1161024-1615-1nkkefz.34qfzjjor/t2quickstart/target/release/build/openssl-sys-a50462f4200b258f/build_script_build" "-Wl,--gc-sections" "-pie" "-Wl,-O1" "-nodefaultlibs" "-L" "/tmp/tessel-rust-compile1161024-1615-1nkkefz.34qfzjjor/t2quickstart/target/release/deps" "-L" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "-Wl,-Bdynamic" "/tmp/tessel-rust-compile1161024-1615-1nkkefz.34qfzjjor/t2quickstart/target/release/deps/libpkg_config-7cc12d9787dada57.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libpanic_unwind-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libunwind-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/librand-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcollections-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_unicode-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/liballoc-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/liballoc_jemalloc-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/liblibc-40393716.rlib" "/root/.multirust/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcore-40393716.rlib" "-l" "dl" "-l" "pthread" "-l" "gcc_s" "-l" "pthread" "-l" "c" "-l" "m" "-l" "rt" "-l" "util" "-l" "compiler-rt"
ERR!
ERR! error: aborting due to previous error
ERR!
ERR! error: Could not compile `openssl-sys`.
ERR!
ERR! To learn more, run the command again with --verbose.
ERR! ERR! "cargo build" exited with status code 101
```

This can be repro'd with:

```
mkdir tesselrepro && cd tesselrepro
t2 init --lang=rust
echo "hyper = \"0.9.x\"" >> Cargo.toml
t2 run blinky --rustcc
```

This likely blocks #56.

felixalias, 2017-02-10:
Bump - has Tessel been abandoned? The last commit was October last year.

I'm honestly a little disappointed - I bought Tessel 1's and had hoped to use them with SSL-secured endpoints. [That didn't pan out, and I was told T2 would address the issue](https://github.com/tessel/t1-runtime/issues/334). So, I bought a bunch of T2s, and I'm met with essentially the same end result. I'm disappointed because T2 really does seem like a much nicer board than competition, but this is a table-stakes feature.

kevinmehall, 2017-02-10:
From the error message, the failure is when it is trying to link the build script for the openssl-sys crate. The build script runs on the host architecture, but the build server doesn't have a native C compiler installed, so Rust can't link binaries that run on the host.

You should open an issue on https://github.com/tessel/rust-compilation-server, or submit a PR there to `apt-get install build-essential` in its Dockerfile.

kevinmehall, 2017-02-10:
Err, it theoretically [has `gcc`](https://github.com/tessel/rust-compilation-server/blob/master/Dockerfile#L4). Maybe an issue with `$PATH`, or something else making it unable to find `cc`.